Xilinx/SLASH #22: QDMA mastering from FPGA

quetric, 2025-05-20:
To support FPGA kernels operating on host-side buffers (and GPU buffers down the line) we need to enable mastering the QDMA from the host side. Modify QDMA config to enable mastering, and implement any required host-side and FPGA-side infrastructure.

lockezhan, 2025-06-27:
Hi @quetric Based on the note "enable mastering the QDMA from the host side", I’d like to confirm whether QDMA is already integrated and actively used, or if this is a planned addition? Thanks

quetric, 2025-06-27:
Hi @lockezhan yes QDMA (in CPM5) is how we access the device over PCIe. Same as AVED.

quetric, 2025-06-27:
At 15 we're running into the maximum number of master ports on an interconnect that Vivado accepts. This can be solved through better BD construction (and also distributing kernels to multiple NOC NMUs) in the linker but we're prioritizing functional testing at the moment and expect to get to these scalability issues soon.

lockezhan, 2025-06-27:
Appreciate the insight, @quetric — that helps clarify the underlying limitation. Totally understand prioritizing functional testing first, and I'm looking forward to seeing how the linker evolves to address scalability in future releases.